siara-cc/esp32-idf-sqlite3 #2: When writing on a SD card, the esp32_Sync function does not flush the data properly

srandoux, 2019-03-04:
Somehow,

the fflush function does not flush data properly and they remain in RAM.
If you change the code to:
int esp32_Sync(sqlite3_file *id, int flags)
{
	esp32_file *file = (esp32_file*) id;

	int rc = fflush( file->fd );
	fsync(fileno(file->fd));
	dbg_printf("esp32_Sync( %s: ): %d \n",file->name, rc);

	return rc ? SQLITE_IOERR_FSYNC : SQLITE_OK;
}

adding, the fsync call, it works as expected.

siara-cc, 2019-03-05:
Thank you.
https://www.esp32.com/viewtopic.php?t=6960
There seems to be some issue about deadlock:
https://github.com/espressif/esp-idf/issues/1837
For now I am including `fsync` as suggested.

srandoux, 2019-03-05:
ok, I'll tell you if experience some deadlocks as described. Thanks.

closed 2019-03-05 by srandoux