siara-cc/esp32-idf-sqlite3 #6: esp-idf-v4.0 support

Elegma, 2020-04-12:
Hello!
How I can build and use this library for esp-idf-v4.0 ?
I try to compile it and have some troubles.

Yardie-, 2021-02-09:
Is there any interest here I too am having compile issues with v4.1

Yardie-, 2021-02-09:
ok I got it working.
I moved the library outside the core directory  ( idf.py complains and fails continually on other projects).
I updated the Makefile adding the appropriate relative path.
then simply used make instead of idf.py
so 
make menuconfig
make flash monitor 
This works on the spiffs example out of the box

Yardie-, 2021-02-10:
After some investigation I found that one difference here is that cmake is now the default 

> CMake-based build system, which is the default since ESP-IDF V4.0. ESP-IDF also supports a legacy build system based on GNU Make, which was the default before ESP-IDF V4.0.

so my guess is that when the source code sqlite3.c was created you used make and now you need to use cmake

siara-cc, 2021-02-14:
@Yardie-  @Elegma I just got it working on esp-idf 4.1.1 and pushed my changes.
Thank you for the valuable inputs.

Yardie-, 2021-02-14:
Wicked well done. You are a legend. I will try it soon. I will also try enabling 8_3 filenames and see if I can get it to work on a virtual fat.

Yardie-, 2021-02-14:
Hi Arun
What was your actual build process please.
Ok I'm on 4.2 stable release
There are some differences but I can't see how it changes the build.
I've tried a couple of things and keep failing.
It's so close I can feel it but I am new to this Cmake stuff and I can't nut it out.

attempt 1 

follow instructions  down load both git repositories
fails 
attempt 2
download and unzip both to ~/esp

cd ~/esp
cp -r ~/esp/esp32-idf-sqlite3-examples-master/spiffs ~/esp
mkdir ~/esp/spiffs/components
cp -r ~/esp/esp32-idf-sqlite3-master ~/esp/spiffs/components
cd ~/esp/spiffs

added to CMakeLists.txt

set(EXTRA_COMPONENT_DIRS ${CMAKE_CURRENT_SOURCE_DIR}/components)

Executing "ninja all"...
[0/1] Re-running CMake...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
CMake Error at /home/g/esp/esp-idf/tools/cmake/component.cmake:221 (message):
  CMake Error at /home/g/esp/spiffs/main/CMakeLists.txt:4
  (idf_build_component):

    Unknown CMake command "idf_build_component".

  Call Stack (most recent call first):

    /home/g/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:60 (include)
    /home/g/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:76 (__component_get_requirements)

Thanks

siara-cc, 2021-02-14:
@Yardie- I have been following this, assuming SDK is installed at `~/esp/esp-idf`:
```
cd ~/esp/esp-idf/components
git clone https://github.com/siara-cc/esp32-idf-sqlite3
cd ../examples
git clone https://github.com/siara-cc/esp32-idf-sqlite3-examples
cd esp32-idf-sqlite3-examples/spiffs
idf.py build
make flash
make monitor
```
After this you should be able to see DBs created and records inserted in SPIFFS.

Yardie-, 2021-02-14:
ok so it still needs to be built with make not cmake 
idf.py menuconfig fails but make menuconfig works 
idf.py flash fails but make flash works

ok that will do for now thanks heaps

closed 2021-06-13 by siara-cc