siara-cc/esp32-idf-sqlite3 #7: Is this lib maintained ?

thbl, 2020-08-15:
Hi @siara-cc 
First off love your work this lib is very cool !

Is the idf version maintained or closed ? ( as i see the Arduino repo has newer code)
Thanks :)

siara-cc, 2020-08-15:
Not really.  I don't really recommend this library or even the [Arduino repo](https://github.com/siara-cc/esp32_arduino_sqlite3_lib) unless you have a specific use case as these are memory hungry.

For logging data into SQLite databases from sensors, I recommend [Sqlite Micro Logger](https://github.com/siara-cc/sqlite_micro_logger_arduino), which leaves a lot of memory free for the developer's use and works with any MCU having 2kb RAM.

thbl, 2020-08-15:
Thans for your input @siara-cc 
I don't need the lib for logging but more for querying so can't use Micro logger as there is no query function sadly.
I need it for a "simple" user  database (id.name, pin...)
with some simple query functions like
get id,name WHERE pin = xxx
update name where id = xxx

Do you have any suggestions ? :) tho this may be a bit off topic

siara-cc, 2020-08-15:
This repo can be used for your purpose. Are you facing any issues?

thbl, 2020-08-16:
Not so far just wanted your input :)
where can i read about the memory usage as you say its hungry ?

siara-cc, 2020-08-16:
https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/7

thbl, 2020-08-17:
Thanks @siara-cc 
Do you have any ide on how i can include this in mongoose os ?
https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md

closed 2020-08-17 by thbl