siara-cc/esp32-idf-sqlite3 #12: Support for Esp32-S3

ArturG16, 2021-11-24:
Hi, I've the new module ESP32-S3.I use this component in my project with ESP32, but I want to update it for the new target.
I changed the Cmakelist with this argument -> set(CMAKE_C_COMPILER "xtensa-esp32s3-elf-gcc").

And worked, but I've got some warnings...
[1029/1312] Building C object esp-idf/..._idf_esp32-idf-sqlite3.dir/esp32.c.obj
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:124:1: warning: missing initializer for field 'xCurrentTimeInt64' of 'sqlite3_vfs' {aka 'struct sqlite3_vfs'} [-Wmissing-field-initializers]
 };
 ^
In file included from /home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:16:
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/include/sqlite3.h:1352:9: note: 'xCurrentTimeInt64' declared here
   int (*xCurrentTimeInt64)(sqlite3_vfs*, sqlite3_int64*);
         ^~~~~~~~~~~~~~~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:140:1: warning: missing initializer for field 'xShmMap' of 'sqlite3_io_methods' {aka 'const struct sqlite3_io_methods'} [-Wmissing-field-initializers]
 };
 ^
In file included from /home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:16:
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/include/sqlite3.h:789:9: note: 'xShmMap' declared here
   int (*xShmMap)(sqlite3_file*, int iPg, int pgsz, int, void volatile**);
         ^~~~~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:156:1: warning: missing initializer for field 'xShmMap' of 'sqlite3_io_methods' {aka 'const struct sqlite3_io_methods'} [-Wmissing-field-initializers]
 };
 ^
In file included from /home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:16:
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/include/sqlite3.h:789:9: note: 'xShmMap' declared here
   int (*xShmMap)(sqlite3_file*, int iPg, int pgsz, int, void volatile**);
         ^~~~~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'esp32mem_Sync':
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:338:14: warning: unused variable 'file' [-Wunused-variable]
  esp32_file *file = (esp32_file*) id;
              ^~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'esp32_Open':
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:393:16: warning: ordered comparison of pointer with integer zero [-Wextra]
     if ( p->fd <= 0 ) {
                ^~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:354:6: warning: unused variable 'rc' [-Wunused-variable]
  int rc;
      ^~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'esp32_Read':
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:430:20: warning: comparison of unsigned expression >= 0 is always true [-Wtype-limits]
  } else if ( nRead >= 0 ) {
                    ^~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'esp32_Truncate':
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:465:14: warning: unused variable 'file' [-Wunused-variable]
  esp32_file *file = (esp32_file*) id;
              ^~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'esp32_Lock':
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:544:14: warning: unused variable 'file' [-Wunused-variable]
  esp32_file *file = (esp32_file*) id;
              ^~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'esp32_Unlock':
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:552:14: warning: unused variable 'file' [-Wunused-variable]
  esp32_file *file = (esp32_file*) id;
              ^~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'esp32_CheckReservedLock':
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:560:14: warning: unused variable 'file' [-Wunused-variable]
  esp32_file *file = (esp32_file*) id;
              ^~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'esp32_FileControl':
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:570:14: warning: unused variable 'file' [-Wunused-variable]
  esp32_file *file = (esp32_file*) id;
              ^~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'esp32_SectorSize':
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:578:14: warning: unused variable 'file' [-Wunused-variable]
  esp32_file *file = (esp32_file*) id;
              ^~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'esp32_DeviceCharacteristics':
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c:586:14: warning: unused variable 'file' [-Wunused-variable]
  esp32_file *file = (esp32_file*) id;
              ^~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/esp32.c: In function 'shox96_0_2d':
[... 296 lines cut ...]
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/sqlite3.c:72141:12: warning: 'btreeHeapPull' defined but not used [-Wunused-function]
 static int btreeHeapPull(u32 *aHeap, u32 *pOut){
            ^~~~~~~~~~~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/sqlite3.c:72131:13: warning: 'btreeHeapInsert' defined but not used [-Wunused-function]
 static void btreeHeapInsert(u32 *aHeap, u32 x){
             ^~~~~~~~~~~~~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/sqlite3.c:65591:20: warning: 'sqlite3BtreeGetAutoVacuum' defined but not used [-Wunused-function]
 SQLITE_PRIVATE int sqlite3BtreeGetAutoVacuum(Btree *p){
                    ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/sqlite3.c:65567:20: warning: 'sqlite3BtreeSetAutoVacuum' defined but not used [-Wunused-function]
 SQLITE_PRIVATE int sqlite3BtreeSetAutoVacuum(Btree *p, int autoVacuum){
                    ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/artur/esp/esp-idf/components/esp32-idf-sqlite3/sqlite3.c:63163:13: warning: 'invalidateAllOverflowCache' defined but not used [-Wunused-function]
 static void invalidateAllOverflowCache(BtShared *pBt){
             ^~~~~~~~~~~~~~~~~~~~~~~~~~
[1311/1312] Generating binary image from built executable
esptool.py v3.2-dev
Merged 2 ELF sections
Generated /home/artur/Projects/blufi/build/blufi_demo.bin
[1312/1312] cd /home/artur/Projects/bl...ur/Projects/blufi/build/blufi_demo.bin
blufi_demo.bin binary size 0xd6700 bytes. Smallest app partition is 0x177000 bytes. 0xa0900 bytes (43%) free.

Project build complete. To flash, run this command:

How can I solved That?

ArturG16, 2021-12-06:
To remove the warnings we must set the new cmake file like that:
set(COMPONENT_ADD_INCLUDEDIRS "include")

idf_component_register(SRCS "sqlite3.c" "esp32.c" "shox96_0_2.c"
                       INCLUDE_DIRS "include"
                       PRIV_INCLUDE_DIRS "private_include"
                       REQUIRES mbedtls
                       PRIV_REQUIRES console spiffs)
set_source_files_properties(
    sqlite3.c 
    PROPERTIES COMPILE_FLAGS
    "-Wno-unused-function -Wno-unused-variable -Wno-unused-value -Wno-unused-label -Wno-missing-field-initializers -Wno-unused-but-set-variable -Wno-maybe-uninitialized -Wno-unused-parameter -Wno-implicit-fallthrough -Wno-cast-function-type"
)
set_source_files_properties(
    esp32.c 
    PROPERTIES COMPILE_FLAGS
       "-Wno-unused-function -Wno-unused-variable -Wno-unused-value -Wno-extra -Wno-type-limits -Wno-unused-label -Wno-missing-field-initializers -Wno-unused-but-set-variable -Wno-maybe-uninitialized -Wno-unused-parameter -Wno-implicit-fallthrough -Wno-cast-function-type"
)
set_source_files_properties(
    shox96_0_2.c 
    PROPERTIES COMPILE_FLAGS
    "-Wno-unused-function -Wno-unused-variable -Wno-char-subscripts -Wno-unused-value -Wno-unused-label -Wno-missing-field-initializers -Wno-unused-but-set-variable -Wno-maybe-uninitialized -Wno-unused-parameter -Wno-implicit-fallthrough -Wno-cast-function-type"
)

target_compile_options(${COMPONENT_LIB} PRIVATE -std=gnu99 -g3 -fno-stack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wpointer-arith -Wno-error=unused-value -Wno-error=unused-label -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-error=char-subscripts -Wno-error=maybe-uninitialized -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -MMD -c -DF_CPU=240000000L -DESP32 -DCORE_DEBUG_LEVEL=0 -DNDEBUG)

ArturG16, 2021-12-06:
And to support all boards we must change on esp32.c

#if CONFIG_IDF_TARGET_ESP32
#include "esp32/rom/ets_sys.h"
#elif CONFIG_IDF_TARGET_ESP32S2
#include "esp32s2/rom/ets_sys.h"
#elif CONFIG_IDF_TARGET_ESP32S3
#include "esp32s3/rom/ets_sys.h"
#endif

siara-cc, 2021-12-07:
Hi @ArturG16  Thank you for sharing the great findings with the newer esp boards.  I will incorporate them in the repo during the weekend.