siara-cc/esp32-idf-sqlite3 #13: cannot use primary key

ankit-thealchemist, 2022-01-23:
Hello, 

I am unable to use the primary key while using on esp-idf v4.3.2. 
While using I am getting an error , sql error: disk i/o error.
I am using the spiffs example.
Thanks

ankit-thealchemist, 2022-01-23:
Hello, 
I found the solution of the problem. 

just change the spiffs config as below. 

esp_vfs_spiffs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = "storage",
    //   .partition_label = NULL,
      .max_files = 5,
      .format_if_mount_failed = true
    };

my partition.csv looks like this 

#### Name,   Type, SubType, Offset,  Size, Flags
##### Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, spiffs,  ,        0x2f0000, 

[image]

closed 2022-01-23 by ankit-thealchemist

JGB2288, 2023-12-02:
I had this problems and much more always with the same error DISK IO Error. or something like that. EVERYTHING SOLVED DONT USING SPIFFS ANYMORE !!!!! Use LittleFS instead !