siara-cc/esp32-idf-sqlite3 #14: how to debug disk I/O error

marchingband, 2023-02-28:
I am using `esp_vfs_fat_sdmmc_mount` to mount on an 8GB eMMC chip, and all was going well.
I am now encountering random `disk I/O error` events when doing `INSERTS`. These occur at random even with a fresh eMMC chip.
I would like to debug what the issue could be, but I don't see a way to turn on debugging in your library.
Can you help? I see other users sharing their debug logs so I know there must be a way.
Thank you!

marchingband, 2023-02-28:
I am also having trouble getting the VERBOSE logs from sqlite3.
If I set the menuconfig log level to VERBOSE, my application crashes (just too many logs).
I have set the menuconfig MAX_LOG_LEVEL to VERBOSE.
I have tried `esp_log_level_set('diskio_sdmmc', ESP_LOG_VERBOSE);`
I have tried editing the CMakeLists.txt and component.mk files in your library to have `-DCORE_DEBUG_LEVEL=5` and removed ` -DNDEBUG`.
but no dice.
Any tips on how to set the component to VERBOSE would be amazing.
Thanks again.

marchingband, 2023-02-28:
i uncommented `#define dbg_printf(...) printf(__VA_ARGS__)`
and got this log:
```
esp32mem_Read: /emmc/tokens.db-journal [45064] [4] OK
esp32_FileControl:
E (26782) diskio_sdmmc: sdmmc_read_blocks failed (257)
esp32_Sync( /emmc/tokens.db: ): 0 
esp32mem_Write: /emmc/tokens.db-journal [0] [28] OK
```

marchingband, 2023-02-28:
Sometimes i also am getting `file is not a database`

marchingband, 2023-03-01:
It turns out my bug was caused by running out of ram.

However I would still like to know how to get verbose logs activated across the library for future.

Thank you!

siara-cc, 2023-03-01:
Are you getting a crash dump? Please see this:
https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/40#issuecomment-765897695

I don't remember, but there are some tools that can show the stack trace using the Backtrace displayed in the crash dump.  This is for Arduino-esp32 I am not sure about esp-idf.

For running out of mem, please consider using Progmem (storing strings in flash) for static strings in your code.

marchingband, 2023-03-01:
I am not getting a crash log because the application doesn't crash. It just returns an error.

siara-cc, 2023-03-01:
I can't think of anything else than what you have figured out about enabling logs.  I think there is a way of enable logs for the Sqlite library using a #define but such logs may not be useful for an already stable library.
Memory issues are difficult to troubleshoot but there is no choice other than slogging to fix them.

closed 2023-03-02 by marchingband