siara-cc/esp32-idf-sqlite3 #22: feature: automatically use psram if available?

elklaaso, 2023-12-08:
by changing this in sqlite.c:

#define SQLITE_MALLOC(x)             malloc(x)
#define SQLITE_FREE(x)               free(x)
#define SQLITE_REALLOC(x,y)          realloc((x),(y))

to:

#ifdef BOARD_HAS_PSRAM
#define SQLITE_MALLOC(x)             ps_malloc(x)
#define SQLITE_FREE(x)               free(x)
#define SQLITE_REALLOC(x,y)          ps_realloc((x),(y))
#else
#define SQLITE_MALLOC(x)             malloc(x)
#define SQLITE_FREE(x)               free(x)
#define SQLITE_REALLOC(x,y)          realloc((x),(y))
#endif

winkelict, 2023-12-14:
great, also did not notice any significant performance degradation on psram btw.

elklaaso, 2023-12-14:
great, also did not notice any significant performance degradation on psram btw.