siara-cc/esp32-idf-sqlite3 #23: Aggregating functions do not work

mciezkowski, 2024-02-21:
Hello
Do the aggregation functions work for you?
I'm using esp32-idf-sqlite3 library on SenseCAP Indicator D1 and I have a problem with aggregate functions.  Normal select works without problem but select with avg, sum does not work. The query is correct because I checked on the PC version of sqlite.  The program does not report an error, just the database does not return anything.

I use callback function and it is a code that returns nothing:
if (!sqlite3_open(DB_PATH, &db11))
            {
            int rc = sqlite3_exec(db11, query_buf1,
                      callback_woda_plot, (void*)data1, &zErrMsg);
            sqlite3_close(db11); 
            }

Have you got any idea what is wrong?
Greetings
Maciek

mciezkowski, 2024-02-21:
Thank you siara-cc,
I will  go follow your suggestions.

Greetings
Maciek

mciezkowski, 2024-02-21:
Hi, 
However, it turns out that the exec function returns error 10 (SQLITE_IOERR */ "disk I/O error",) when the query contains an aggregate function. 
Does this somehow make the problem easier to diagnose?
I use spiffs.

Greetings
Maciek

siara-cc, 2024-02-22:
SPIFFS known to cause such problems. Can try LITTLEFS?
https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/18

mciezkowski, 2024-02-22:
Thanks  a lot! 
This is highly likely.

Greetings
Maciek