siara-cc/esp32-idf-sqlite3 #32: Error Compilation ESP-IDF 5.3.1

hmaLiiAtech, 2024-09-12:
Hi,
I use Linux Ubuntu & ESP-IDF 5.3.1
I get error when build : 

/home/liia/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj):(.literal.app_main+0x84): undefined reference to `sqlite3_open'
/home/liia/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj): in function `app_main':
/home/liia/sac_nw_v10/main/main.cpp:238:(.text.app_main+0x1e8): undefined reference to `sqlite3_open'

Code 👍 

#include <stdio.h>
#include "sqlite3.h"

extern "C" void app_main(void) {
    sqlite3 *db;
    int rc = sqlite3_open(":memory:", &db);
    if (rc) {
        printf("Can't open database: %s\n", sqlite3_errmsg(db));
    } else {
        printf("Opened database successfully\n");
    }
    sqlite3_close(db);
}

sqlite3.h is a component in the folder components/sqlite

CMakeLists.txt :  (in the sqlite)

set(COMPONENT_SRCDIRS ".")
set(COMPONENT_ADD_INCLUDEDIRS ".")
set(COMPONENT_REQUIRES console spi_flash esp_partition esp_timer spiffs json)
register_component()

CMakeLists.txt (in main)

set(COMPONENT_SRCDIRS ".")
set(COMPONENT_ADD_INCLUDEDIRS ".")

register_component()

this configuration works in V4.4.6.
it was used but on mac OSX.

thanks for your support

nopnop2002, 2024-11-06:
The repository that can be built with ESP-IDF V5.X is available here.

https://github.com/nopnop2002/esp32-idf-sqlite3

siara-cc, 2024-11-06:
@nopnop2002  Thank you for sharing your work!

Zhu-jiatong, 2025-01-14:
@siara-cc is this repo still maintained? or are we switching to [this repo](https://github.com/nopnop2002/esp32-idf-sqlite3)? Thanks.