siara-cc/esp32-idf-sqlite3 #36: ESP32 nodemcu-32s - "Can't open database: unable to open database file"

pixel-shock, 2025-01-20:
Hi,

I am getting this error while following the examples 

`Can't open database: unable to open database file`

and have no clue why. The file exists on the FS (SPIFFS) and it's accessable and readable with "normal" FS functions like `fs.open`.

[image]

I would appreciate any ideas to solve the problem 😉

nopnop2002, 2025-01-24:
Normally, logging would look something like this.
Your logging does not include boot logging.
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (31) boot: ESP-IDF v5.3.2-1-g9a6aba7401-dirty 2nd stage bootloader
I (31) boot: compile time Jan 25 2025 06:34:59
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v1.0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (100) boot: End of partition table
```

pixel-shock, 2025-01-25:
sorry for that ... that was just the only log I got.
But nontheless - I switchted to a simple json storage ... the lib is to big for my project and it's not working and maybe not maintained anymore ..

But thank you anyways!